allenkey666/openwrt-kcptun #2: 6/27编译的版本运行起来有错误

JogFeelingVI, 2016-06-27:
2016/06/27 09:06:16 version: SELFBUILD
2016/06/27 09:06:17 listening on: 0.0.0.0:2000
2016/06/27 09:06:17 encryption: aes
2016/06/27 09:06:17 nodelay parameters: 1 20 2 1
2016/06/27 09:06:17 remote address: 104.128.80.170:29900
2016/06/27 09:06:17 sndwnd: 128 rcvwnd: 1024
2016/06/27 09:06:17 compression: true
2016/06/27 09:06:17 mtu: 1350
2016/06/27 09:06:17 datashard: 10 parityshard: 3
2016/06/27 09:06:17 acknodelay: false
2016/06/27 09:06:17 dscp: 0
2016/06/27 09:06:17 conn: 1
2016/06/27 09:06:54 stream opened
panic: runtime error: slice bounds out of range

goroutine 11 [running]:
github.com/golang/snappy.encodeBlock(0x105ae013, 0x12ac9, 0x12ac9, 0x1059e000, 0x17, 0x10000, 0x53a95f61)
	/opt/mipsgo/src/github.com/golang/snappy/encode_other.go:175 +0x10c4
github.com/golang/snappy.Encode(0x105ae012, 0x12aca, 0x12aca, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:37 +0x370
github.com/golang/snappy.(*Writer).write(0x1053c690, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:229 +0x280
github.com/golang/snappy.(*Writer).Flush(0x1053c690, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:272 +0xb0
main.(*compStream).Write(0x10538130, 0x1060a000, 0x17, 0x8000, 0x17, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:37 +0x88
bytes.(*Reader).WriteTo(0x10536e00, 0x76ff6440, 0x10538130, 0x76ff6458, 0x10536e00, 0x0, 0x0)
	/opt/mipsgo/src/bytes/reader.go:131 +0x13c
io.Copy(0x76ff6440, 0x10538130, 0x76ff6428, 0x10536e00, 0x0, 0x0, 0x0, 0x0)

closed 2016-06-27 by JogFeelingVI

bettermanbao, 2016-06-28:
错误解决了？

JogFeelingVI, 2016-06-29:
yes 原因是 客户端和服务器端 没有加 -ncomp的缘故~

andres-lee, 2016-07-03:
6月27日的版本开启压缩是乎是有问题的，不知是否是snappy里出啥问题了。
```
2015/06/05 05:13:35 conn: 1
2015/06/05 05:13:37 stream opened
panic: runtime error: slice bounds out of range

goroutine 11 [running]:
github.com/golang/snappy.encodeBlock(0x105ae014, 0x12ac8, 0x12ac8, 0x10582000, 0xe9, 0x10000, 0x56e922bd)
	/opt/mipsgo/src/github.com/golang/snappy/encode_other.go:175 +0x10c4
github.com/golang/snappy.Encode(0x105ae012, 0x12aca, 0x12aca, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:37 +0x370
github.com/golang/snappy.(*Writer).write(0x1053c690, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:229 +0x280
github.com/golang/snappy.(*Writer).Flush(0x1053c690, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:272 +0xb0
main.(*compStream).Write(0x1053a650, 0x10602000, 0xe9, 0x8000, 0xe9, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:37 +0x88
bytes.(*Reader).WriteTo(0x10538ac0, 0x77484460, 0x1053a650, 0x77484478, 0x10538ac0, 0x0, 0x0)
	/opt/mipsgo/src/bytes/reader.go:131 +0x13c
io.Copy(0x77484460, 0x1053a650, 0x77484448, 0x10538ac0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:354 +0xac
github.com/hashicorp/yamux.(*Session).send(0x1053e000)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:387 +0x34c
created by github.com/hashicorp/yamux.newSession
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:109 +0x5c0
```

bettermanbao, 2016-07-04:
是有问题，但是看不懂问题在哪里。凑活一下用-nocomp参数启动，禁用压缩吧。反正mips32本身性能就不行，就不要开启压缩了。

andres-lee, 2016-07-04:
嗯嗯，没啥影响。 : )