allenkey666/openwrt-kcptun #3: 编译教程

saginardo, 2016-06-30:
你好，大神能不能出个简易的编译教程啊。

hbxntsxw, 2016-06-30:
我也想学习一下自己编译！

saginardo, 2016-06-30:
我试着google了下 golang的mips32的编译，发现github有个gomini的，但是编译报错不成功，搜一下也没答案；不知道是不是缺什么库来着

bettermanbao, 2016-06-30:
github 搜 go mips32

saginardo, 2016-06-30:
我就是搜的这个，但是编译按照步骤好像有错，最后在bin目录下面也没有生成go的执行文件
类似于这种错误
unknown relocation type 42; compiled without -fpic
unexpected relocation type 298

saginardo, 2016-06-30:
看了一个网上搜的教程
git clone https://github.com/gomini/go-mips32.git 
cd go-mips32/src
//配置GO编译参数 
export GOOS=linux 
export GOARCH=mips32le <== Change to mips32 if mips
//执行编译
./make.bash 
cd ..
//创建编译后文件存放文件夹 
sudo mkdir /opt/mipsgo
//复制 
sudo cp -R * /opt/mipsgo
//go工程参数配置 
export GOROOT=/opt/mipsgo 
export PATH=/opt/mipsgo/bin:$PATH

我查看了下 /opt/mipsgo/bin下只有 gofmt，这样是不是没有编译成功啊；运行go build也提示未找到go，希望大神指导下是不是漏掉什么步骤了。

saginardo, 2016-06-30:
编译日志如下
# Building C bootstrap tool.
cmd/dist

# Building compilers and Go bootstrap tool for host, linux/amd64.
lib9
libbio
liblink
cmd/cc
cmd/gc
cmd/6l
cmd/vl
cmd/6a
cmd/va
cmd/6c
cmd/vc
cmd/6g
cmd/vg
runtime
errors
sync/atomic
sync
io
unicode
unicode/utf8
unicode/utf16
bytes
math
strings
strconv
bufio
sort
container/heap
encoding/base64
syscall
time
os
reflect
fmt
encoding
encoding/json
flag
path/filepath
path
io/ioutil
log
regexp/syntax
regexp
go/token
go/scanner
go/ast
go/parser
os/exec
os/signal
net/url
text/template/parse
text/template
go/doc
go/build
cmd/go
runtime (linux/mips32le)

# Building packages and commands for host, linux/amd64.
runtime
errors
sync/atomic
sync
io
unicode
unicode/utf8
bytes
bufio
math
syscall
time
os
strconv
reflect
fmt
strings
cmd/internal/goobj
encoding/binary
cmd/internal/rsc.io/arm/armasm
cmd/internal/rsc.io/x86/x86asm
debug/dwarf
debug/elf
debug/gosym
debug/macho
debug/pe
debug/plan9obj
sort
regexp/syntax
regexp
text/tabwriter
cmd/internal/objfile
flag
log
cmd/addr2line
hash
crypto
crypto/md5
go/token
path/filepath
go/scanner
go/ast
io/ioutil
go/parser
go/printer
os/exec
cmd/cgo
go/format
path
cmd/fix
container/heap
encoding
encoding/base64
unicode/utf16
encoding/json
encoding/xml
net/url
text/template/parse
text/template
go/doc
go/build
compress/flate
hash/crc32
compress/gzip
container/list
crypto/subtle
crypto/cipher
crypto/aes
crypto/des
math/rand
math/big
crypto/elliptic
encoding/asn1
crypto/ecdsa
crypto/hmac
internal/syscall
crypto/rand
crypto/rc4
crypto/rsa
crypto/sha1
crypto/sha256
crypto/dsa
crypto/sha512
crypto/x509/pkix
encoding/hex
encoding/pem
runtime/cgo
net
crypto/x509
crypto/tls
mime
net/textproto
mime/multipart
net/http/internal
net/http
os/signal
cmd/go
# cmd/go
/home/sag/go-mips32/pkg/linux_amd64/runtime/cgo.a(_all.o): unknown relocation type 42; compiled without -fpic?
/home/sag/go-mips32/pkg/linux_amd64/runtime/cgo.a(_all.o): unknown relocation type 42; compiled without -fpic?
runtime/cgo(.text): unexpected relocation type 298
runtime/cgo(.text): unexpected relocation type 298
runtime/pprof
cmd/gofmt
cmd/nm
cmd/objdump
cmd/pack
cmd/pprof/internal/profile
cmd/pprof/internal/plugin
html
html/template
cmd/pprof/internal/report
cmd/pprof/internal/svg
cmd/pprof/internal/tempfile
cmd/pprof/internal/commands
cmd/pprof/internal/driver
cmd/pprof/internal/fetch
cmd/pprof/internal/symbolizer
cmd/pprof/internal/symbolz
cmd/pprof
# cmd/pprof
/home/sag/go-mips32/pkg/linux_amd64/runtime/cgo.a(_all.o): unknown relocation type 42; compiled without -fpic?
/home/sag/go-mips32/pkg/linux_amd64/runtime/cgo.a(_all.o): unknown relocation type 42; compiled without -fpic?
runtime/cgo(.text): unexpected relocation type 298
runtime/cgo(.text): unexpected relocation type 298
cmd/yacc
archive/tar
archive/zip
compress/bzip2
compress/lzw
hash/adler32
compress/zlib
container/ring
database/sql/driver
database/sql
encoding/ascii85
encoding/base32
encoding/csv
encoding/gob
expvar
hash/crc64
hash/fnv
image/color
image
image/color/palette
image/draw
image/gif
image/jpeg
image/png
index/suffixarray
log/syslog
math/cmplx
net/http/cgi
net/http/cookiejar
net/http/fcgi
net/http/httptest
net/http/httputil
net/http/pprof
net/mail
net/rpc
net/rpc/jsonrpc
net/smtp
os/user
runtime/debug
runtime/race
testing
testing/iotest
testing/quick
text/scanner

bettermanbao, 2016-06-30:
这个教程就是我写的

saginardo, 2016-06-30:
额，这样啊；那你知道我这个是哪里出问题了么

wang20150419, 2016-07-12:
kcptun 有 20160701版本了，期待更新对应的 openwrt版本。

closed 2016-07-14 by saginardo