allenkey666/openwrt-kcptun #5: 开启压缩模式后出错退出

sbugzu, 2016-08-03:
出错后把服务器端升级到最新版本仍然报错，加 -nocomp 能正常运行。
版本信息如下：

```
OpenWrt Chaos Calmer 15.05
kcptun-linux-amd64-20160725
kcptun_client_openwrt_ar71xx_20160701
```

出错信息：

```
version: SELFBUILD
2016/08/03 07:31:01 listening on: [::]:41080
2016/08/03 07:31:01 encryption: aes
2016/08/03 07:31:01 nodelay parameters: 1 20 2 1
2016/08/03 07:31:01 remote address: xxx.xxx.xxx.xxx:xxxx
2016/08/03 07:31:01 sndwnd: 256 rcvwnd: 512
2016/08/03 07:31:01 compression: true
2016/08/03 07:31:01 mtu: 1400
2016/08/03 07:31:01 datashard: 10 parityshard: 3
2016/08/03 07:31:01 acknodelay: false
2016/08/03 07:31:01 dscp: 46
2016/08/03 07:31:01 conn: 1
2016/08/03 07:31:01 stream opened
panic: runtime error: slice bounds out of range

goroutine 11 [running]:
github.com/golang/snappy.encodeBlock(0x105ae014, 0x12ac8, 0x12ac8, 0x1059e000, 0x21c, 0x10000, 0x542aa73)
        /opt/mipsgo/src/github.com/golang/snappy/encode_other.go:175 +0x10c4
github.com/golang/snappy.Encode(0x105ae012, 0x12aca, 0x12aca, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/github.com/golang/snappy/encode.go:37 +0x370
github.com/golang/snappy.(*Writer).write(0x1053e6c0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/github.com/golang/snappy/encode.go:229 +0x280
github.com/golang/snappy.(*Writer).Flush(0x1053e6c0, 0x0, 0x0)
        /opt/mipsgo/src/github.com/golang/snappy/encode.go:272 +0xb0
main.(*compStream).Write(0x1053a230, 0x105f4000, 0x21c, 0x8000, 0x21c, 0x0, 0x0)
        /home/openwrt/kcptun/client/main.go:39 +0x88
bytes.(*Reader).WriteTo(0x10538aa0, 0x77416530, 0x1053a230, 0x77416548, 0x10538aa0, 0x0, 0x0)
        /opt/mipsgo/src/bytes/reader.go:131 +0x13c
io.Copy(0x77416530, 0x1053a230, 0x77416518, 0x10538aa0, 0x0, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:354 +0xac
github.com/hashicorp/yamux.(*Session).send(0x1050c070)
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:387 +0x34c
created by github.com/hashicorp/yamux.newSession
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:109 +0x5c0

goroutine 1 [IO wait]:
net.(*pollDesc).Wait(0x1054a178, 0x72, 0x0, 0x0)
        /opt/mipsgo/src/net/fd_poll_runtime.go:84 +0x4c
net.(*pollDesc).WaitRead(0x1054a178, 0x0, 0x0)
        /opt/mipsgo/src/net/fd_poll_runtime.go:89 +0x48
net.(*netFD).accept(0x1054a140, 0x0, 0x77414ad8, 0x1053a2b8)
        /opt/mipsgo/src/net/fd_unix.go:419 +0x3dc
net.(*TCPListener).AcceptTCP(0x1051c7b8, 0x1051c2a8, 0x0, 0x0)
        /opt/mipsgo/src/net/tcpsock_posix.go:234 +0x58
main.func·004(0x10500820, 0x0, 0x0)
        /home/openwrt/kcptun/client/main.go:267 +0x19ac
reflect.Value.call(0x1f1840, 0x288a80, 0x13, 0x23da48, 0x4, 0x10529b6c, 0x1, 0x1, 0x0, 0x0, ...)
        /opt/mipsgo/src/reflect/value.go:419 +0xff0
reflect.Value.Call(0x1f1840, 0x288a80, 0x13, 0x10529b6c, 0x1, 0x1, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/reflect/value.go:296 +0xb0
github.com/urfave/cli.HandleAction(0x1f1840, 0x288a80, 0x10500820, 0x0, 0x0)
        /opt/mipsgo/src/github.com/urfave/cli/app.go:485 +0x27c
github.com/urfave/cli.(*App).Run(0x1055c000, 0x1050a000, 0xf, 0xf, 0x0, 0x0)
        /opt/mipsgo/src/github.com/urfave/cli/app.go:244 +0x9e0
main.main()
        /home/openwrt/kcptun/client/main.go:283 +0xff4

goroutine 5 [select]:
github.com/xtaci/kcp-go.(*UDPSession).updateTask(0x1053c120)
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:429 +0x2e0
created by github.com/xtaci/kcp-go.newUDPSession
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:104 +0x5fc

goroutine 6 [runnable]:
github.com/xtaci/kcp-go.(*UDPSession).outputTask(0x1053c120)
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:336 +0x1804
created by github.com/xtaci/kcp-go.newUDPSession
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:105 +0x62c

goroutine 7 [select]:
github.com/xtaci/kcp-go.(*UDPSession).readLoop(0x1053c120)
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:535 +0x578
created by github.com/xtaci/kcp-go.newUDPSession
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:107 +0x668

goroutine 9 [IO wait]:
net.(*pollDesc).Wait(0x1054aff8, 0x72, 0x0, 0x0)
        /opt/mipsgo/src/net/fd_poll_runtime.go:84 +0x4c
net.(*pollDesc).WaitRead(0x1054aff8, 0x0, 0x0)
        /opt/mipsgo/src/net/fd_poll_runtime.go:89 +0x48
net.(*netFD).readFrom(0x1054afc0, 0x10584800, 0x800, 0x800, 0x0, 0x0, 0x0, 0x77414ad8, 0x1053a260)
        /opt/mipsgo/src/net/fd_unix.go:269 +0x418
net.(*UDPConn).ReadFromUDP(0x1051c3f0, 0x10584800, 0x800, 0x800, 0x800, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/net/udpsock_posix.go:67 +0x108
github.com/xtaci/kcp-go.(*UDPSession).receiver(0x1053c120, 0x1054a280)
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:516 +0x108
created by github.com/xtaci/kcp-go.(*UDPSession).readLoop
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:532 +0x84

goroutine 10 [select]:
github.com/xtaci/kcp-go.(*UDPSession).Read(0x1053c120, 0x105d2000, 0x4, 0x12ace, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:169 +0x854
io.ReadAtLeast(0x77416348, 0x1053c120, 0x105d2000, 0x4, 0x12ace, 0x4, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:298 +0x118
io.ReadFull(0x77416348, 0x1053c120, 0x105d2000, 0x4, 0x12ace, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:316 +0x6c
github.com/golang/snappy.(*Reader).readFull(0x1054a2c0, 0x105d2000, 0x4, 0x12ace, 0x1000000, 0x0)
        /opt/mipsgo/src/github.com/golang/snappy/decode.go:108 +0x6c
github.com/golang/snappy.(*Reader).Read(0x1054a2c0, 0x10548000, 0x1000, 0x1000, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/github.com/golang/snappy/decode.go:128 +0x1bc
main.(*compStream).Read(0x1053a230, 0x10548000, 0x1000, 0x1000, 0x0, 0x0, 0x0)
        /home/openwrt/kcptun/client/main.go:34 +0x5c
bufio.(*Reader).fill(0x1053e720)
        /opt/mipsgo/src/bufio/bufio.go:97 +0x1fc
bufio.(*Reader).Read(0x1053e720, 0x1053a250, 0xc, 0xc, 0x1, 0x0, 0x0)
        /opt/mipsgo/src/bufio/bufio.go:174 +0x25c
io.ReadAtLeast(0x77416398, 0x1053e720, 0x1053a250, 0xc, 0xc, 0xc, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:298 +0x118
io.ReadFull(0x77416398, 0x1053e720, 0x1053a250, 0xc, 0xc, 0x1053a250, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:316 +0x6c
github.com/hashicorp/yamux.(*Session).recvLoop(0x1050c070, 0x0, 0x0)
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:418 +0x14c
github.com/hashicorp/yamux.(*Session).recv(0x1050c070)
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:406 +0x38
created by github.com/hashicorp/yamux.newSession
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:108 +0x594

goroutine 12 [select]:
github.com/hashicorp/yamux.(*Session).keepalive(0x1050c070)
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:302 +0x23c
created by github.com/hashicorp/yamux.newSession
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:111 +0x5fc

goroutine 13 [select]:
main.handleClient(0x774163e8, 0x1051c2a8, 0x77416408, 0x1050a080)
        /home/openwrt/kcptun/client/main.go:75 +0x65c
created by main.func·004
        /home/openwrt/kcptun/client/main.go:278 +0x1cdc

goroutine 14 [select]:
github.com/hashicorp/yamux.(*Stream).Read(0x1050a080, 0x105ec000, 0x8000, 0x8000, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/github.com/hashicorp/yamux/stream.go:125 +0x458
io.Copy(0x77416488, 0x1051c3a8, 0x77416440, 0x1050a080, 0x0, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:362 +0x208
net.genericReadFrom(0x77416428, 0x1051c2a8, 0x77416440, 0x1050a080, 0x0, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/net/net.go:404 +0xb4
net.(*TCPConn).ReadFrom(0x1051c2a8, 0x77416440, 0x1050a080, 0x77416470, 0x1051c2a8, 0x0, 0x0)
        /opt/mipsgo/src/net/tcpsock_posix.go:72 +0x118
io.Copy(0x77416428, 0x1051c2a8, 0x77416440, 0x1050a080, 0x0, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:358 +0x154
main.func·001()
        /home/openwrt/kcptun/client/main.go:64 +0xd8
created by main.handleClient
        /home/openwrt/kcptun/client/main.go:66 +0x448

goroutine 15 [select]:
github.com/hashicorp/yamux.(*Session).waitForSendErr(0x1050c070, 0x1053a120, 0xc, 0xc, 0x77416518, 0x10538aa0, 0x1054a580, 0x0, 0x0)
        /opt/mipsgo/src/github.com/hashicorp/yamux/session.go:338 +0x330
github.com/hashicorp/yamux.(*Stream).write(0x1050a080, 0x105f4000, 0x21c, 0x8000, 0x61250, 0x0, 0x0)
        /opt/mipsgo/src/github.com/hashicorp/yamux/stream.go:183 +0x594
github.com/hashicorp/yamux.(*Stream).Write(0x1050a080, 0x105f4000, 0x21c, 0x8000, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/github.com/hashicorp/yamux/stream.go:139 +0x158
io.Copy(0x774164b8, 0x1050a080, 0x774164d0, 0x1051c2a8, 0x0, 0x0, 0x0, 0x0)
        /opt/mipsgo/src/io/io.go:364 +0x288
main.func·002()
        /home/openwrt/kcptun/client/main.go:70 +0xd8
created by main.handleClient
        /home/openwrt/kcptun/client/main.go:72 +0x578
```

bettermanbao, 2016-08-03:
snappy的压缩在mips32下确实有问题，我我也没本事修，只能禁用了。

sbugzu, 2016-08-04:
了解，先禁用吧，感谢作者的编译。

closed 2016-09-21 by bettermanbao