allenkey666/openwrt-kcptun #6: AR9341路由器用KCP速度太慢了，KCP客户端进程还经常被kill

ShureCool, 2016-08-07:
AR9341路由器刷Openwrt之后，运行KCP客户端，同时SSR配置本地KCP的端口，速度很慢，KCP客户端进程还经常被kill

bettermanbao, 2016-08-08:
确实如此，所以我最终还是买了个山寨的RaspberryPi - Orange Pi，单独运行arm版的kcptun。

xayoung, 2016-09-02:
@bettermanbao 这个方案挺不错的啊，我的3800路由器估计也跑不了你编译的kcptun，望指导下RaspberryPi跑kcptun方案。

youuoypi, 2016-09-29:
Orange pi是怎么加速?运行arm版本的kcptun,给路由器加速?路由器和Orange Pi是怎么连接的?

bettermanbao, 2016-09-30:
用raspberry pi 或 orange pi做本地中转，然后路由器的ss服务器地址指向pi，就可以了。

closed 2016-09-30 by bettermanbao

youuoypi, 2016-09-30:
pi接的是路由器的lan口吗?

bettermanbao, 2016-09-30:
是的

ninggu2008, 2016-11-29:
@bettermanbao 买哪个版本的orange pi？好像有大中小三种规格？四核A7的，四核A53的？性能哪个够用？

bettermanbao, 2016-11-29:
@ninggu2008 最新版的kcptun已经可以在ar9341的路由器上流畅运行了。
orangepi买最便宜的就可以了，orangepi zero 4核关掉3核依旧性能过剩。

ninggu2008, 2016-12-02:
@bettermanbao 我是HG255D，RT3052F的CPU。非常老了。主要是多拨，ss-local，kcp。ss的加密方式rc4-md5，kcp是salsa20。好像负载是不高（2000kbps下，看480p），从ssh看，负载保持在不到1的水平。kcp工作时占用cpu有50%-80%。ss工作时占用却很低。现在是冬天，路由器倒是不怎么怕发热。还有个二级路由，小米路由mini，7620的cpu，运行着adbyby，偶尔离线下个美剧，无线设备也都连到上面去，我感觉用二级运行kcptun，还不如1级的，1级关了无线功能应该任务轻点。