allenkey666/openwrt-kcptun #8: panic: runtime error: slice bounds out of range

MaiReo, 2016-08-13:
goroutine 11 [running]:
github.com/golang/snappy.encodeBlock(0x105ae014, 0x12ac8, 0x12ac8, 0x10582000, 0x218, 0x10000, 0x432ec0b7)
	/opt/mipsgo/src/github.com/golang/snappy/encode_other.go:175 +0x10c4
github.com/golang/snappy.Encode(0x105ae012, 0x12aca, 0x12aca, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:37 +0x370
github.com/golang/snappy.(*Writer).write(0x1053e6f0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:229 +0x280
github.com/golang/snappy.(*Writer).Flush(0x1053e6f0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:272 +0xb0
main.(*compStream).Write(0x1053a640, 0x10610000, 0x218, 0x8000, 0x218, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:39 +0x88
bytes.(*Reader).WriteTo(0x10538c60, 0x776a4470, 0x1053a640, 0x776a4488, 0x10538c60, 0x0, 0x0)
	/opt/mipsgo/src/bytes/reader.go:131 +0x13c
io.Copy(0x776a4470, 0x1053a640, 0x776a4458, 0x10538c60, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:354 +0xac
github.com/hashicorp/yamux.(*Session).send(0x1050c070)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:387 +0x34c
created by github.com/hashicorp/yamux.newSession
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:109 +0x5c0

goroutine 1 [select]:
github.com/hashicorp/yamux.(*Session).waitForSendErr(0x1050c070, 0x1053a8e0, 0xc, 0xc, 0x0, 0x0, 0x1054b400, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:338 +0x330
github.com/hashicorp/yamux.(*Stream).sendWindowUpdate(0x1050a200, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/stream.go:256 +0x240
github.com/hashicorp/yamux.(*Session).OpenStream(0x1050c070, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:177 +0x34c
github.com/hashicorp/yamux.(*Session).Open(0x1050c070, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:136 +0x48
main.func·004(0x10500960, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:305 +0x20cc
reflect.Value.call(0x1f18d8, 0x2896a8, 0x13, 0x23dfa0, 0x4, 0x1052fb24, 0x1, 0x1, 0x0, 0x0, ...)
	/opt/mipsgo/src/reflect/value.go:419 +0xff0
reflect.Value.Call(0x1f18d8, 0x2896a8, 0x13, 0x1052fb24, 0x1, 0x1, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/reflect/value.go:296 +0xb0
github.com/urfave/cli.HandleAction(0x1f18d8, 0x2896a8, 0x10500960, 0x0, 0x0)
	/opt/mipsgo/src/github.com/urfave/cli/app.go:485 +0x27c
github.com/urfave/cli.(*App).Run(0x1055c000, 0x1050a000, 0xf, 0xf, 0x0, 0x0)
	/opt/mipsgo/src/github.com/urfave/cli/app.go:244 +0x9e0
main.main()
	/home/openwrt/kcptun/client/main.go:317 +0x11cc

goroutine 5 [select]:
github.com/xtaci/kcp-go.(*UDPSession).updateTask(0x10500780)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:488 +0x2e0
created by github.com/xtaci/kcp-go.newUDPSession
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:106 +0x614

goroutine 6 [runnable]:
github.com/xtaci/kcp-go.(*UDPSession).outputTask(0x10500780)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:387 +0x1aa8
created by github.com/xtaci/kcp-go.newUDPSession
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:107 +0x644

goroutine 7 [select]:
github.com/xtaci/kcp-go.(*UDPSession).readLoop(0x10500780)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:594 +0x578
created by github.com/xtaci/kcp-go.newUDPSession
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:109 +0x680

goroutine 9 [IO wait]:
net.(*pollDesc).Wait(0x1054aff8, 0x72, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:84 +0x4c
net.(*pollDesc).WaitRead(0x1054aff8, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:89 +0x48
net.(*netFD).Read(0x1054afc0, 0x10594800, 0x800, 0x800, 0x0, 0x776a2ad8, 0x1053a744)
	/opt/mipsgo/src/net/fd_unix.go:242 +0x3e0
net.(*conn).Read(0x1051c3e8, 0x10594800, 0x800, 0x800, 0x4b1, 0x0, 0x0)
	/opt/mipsgo/src/net/net.go:121 +0xf0
github.com/xtaci/kcp-go.(*UDPSession).receiver(0x10500780, 0x1054a280)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:575 +0x114
created by github.com/xtaci/kcp-go.(*UDPSession).readLoop
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:591 +0x84

goroutine 10 [select]:
github.com/xtaci/kcp-go.(*UDPSession).Read(0x10500780, 0x105d2000, 0x4, 0x12ace, 0xc, 0x0, 0x0)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:171 +0x854
io.ReadAtLeast(0x776a4210, 0x10500780, 0x105d2000, 0x4, 0x12ace, 0x4, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:298 +0x118
io.ReadFull(0x776a4210, 0x10500780, 0x105d2000, 0x4, 0x12ace, 0x1054c800, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:316 +0x6c
github.com/golang/snappy.(*Reader).readFull(0x1054a2c0, 0x105d2000, 0x4, 0x12ace, 0x1, 0x1)
	/opt/mipsgo/src/github.com/golang/snappy/decode.go:108 +0x6c
github.com/golang/snappy.(*Reader).Read(0x1054a2c0, 0x10548000, 0x1000, 0x1000, 0xc, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/decode.go:128 +0x1bc
main.(*compStream).Read(0x1053a640, 0x10548000, 0x1000, 0x1000, 0xc, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:34 +0x5c
bufio.(*Reader).fill(0x1053e750)
	/opt/mipsgo/src/bufio/bufio.go:97 +0x1fc
bufio.(*Reader).Read(0x1053e750, 0x1053a150, 0xc, 0xc, 0x105388c0, 0x0, 0x0)
	/opt/mipsgo/src/bufio/bufio.go:174 +0x25c
io.ReadAtLeast(0x776a42d8, 0x1053e750, 0x1053a150, 0xc, 0xc, 0xc, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:298 +0x118
io.ReadFull(0x776a42d8, 0x1053e750, 0x1053a150, 0xc, 0xc, 0xc, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:316 +0x6c
github.com/hashicorp/yamux.(*Session).recvLoop(0x1050c070, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:418 +0x14c
github.com/hashicorp/yamux.(*Session).recv(0x1050c070)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:406 +0x38
created by github.com/hashicorp/yamux.newSession
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:108 +0x594

goroutine 12 [select]:
github.com/hashicorp/yamux.(*Session).keepalive(0x1050c070)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:302 +0x23c
created by github.com/hashicorp/yamux.newSession
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:111 +0x5fc

goroutine 14 [select]:
main.handleClient(0x776a4328, 0x1051c398, 0x776a4348, 0x1050a100)
	/home/openwrt/kcptun/client/main.go:75 +0x65c
created by main.func·004
	/home/openwrt/kcptun/client/main.go:313 +0x2348

goroutine 15 [select]:
github.com/hashicorp/yamux.(*Stream).Read(0x1050a100, 0x10608000, 0x8000, 0x8000, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/stream.go:130 +0x4e0
io.Copy(0x776a43c8, 0x1051c428, 0x776a4380, 0x1050a100, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:362 +0x208
net.genericReadFrom(0x776a4368, 0x1051c398, 0x776a4380, 0x1050a100, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/net/net.go:404 +0xb4
net.(*TCPConn).ReadFrom(0x1051c398, 0x776a4380, 0x1050a100, 0x776a43b0, 0x1051c398, 0x0, 0x0)
	/opt/mipsgo/src/net/tcpsock_posix.go:72 +0x118
io.Copy(0x776a4368, 0x1051c398, 0x776a4380, 0x1050a100, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:358 +0x154
main.func·001()
	/home/openwrt/kcptun/client/main.go:64 +0xd8
created by main.handleClient
	/home/openwrt/kcptun/client/main.go:66 +0x448

goroutine 16 [select]:
github.com/hashicorp/yamux.(*Session).waitForSendErr(0x1050c070, 0x1053a790, 0xc, 0xc, 0x776a4458, 0x10538c60, 0x1054ad80, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/session.go:338 +0x330
github.com/hashicorp/yamux.(*Stream).write(0x1050a100, 0x10610000, 0x218, 0x8000, 0x61ad8, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/stream.go:191 +0x594
github.com/hashicorp/yamux.(*Stream).Write(0x1050a100, 0x10610000, 0x218, 0x8000, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/hashicorp/yamux/stream.go:147 +0x158
io.Copy(0x776a43f8, 0x1050a100, 0x776a4410, 0x1051c398, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:364 +0x288
main.func·002()
	/home/openwrt/kcptun/client/main.go:70 +0xd8
created by main.handleClient
	/home/openwrt/kcptun/client/main.go:72 +0x578

goroutine 17 [runnable]:
main.handleClient(0x776a4328, 0x1051c3b8, 0x776a4348, 0x1050a180)
	/home/openwrt/kcptun/client/main.go:55
created by main.func·004
	/home/openwrt/kcptun/client/main.go:313 +0x2348

MaiReo, 2016-08-13:
是哪里设置的不对吗

fy0, 2016-10-17:
@MaiReo   -nocomp就可以了 应该是压缩的BUG

ljyloo, 2016-10-25:
@fy0 是的！关闭压缩就好了，感谢！

MaiReo, 2016-10-25:
@fy0 感谢

closed 2016-10-25 by MaiReo

jemyzhang, 2016-12-22:
我也碰到同样的问题, 这个不算bug? 只能关压缩么? @bettermanbao

bettermanbao, 2016-12-22:
算bug吧