allenkey666/openwrt-kcptun #14: WR841N，openwrt上有看到stream产生，但实际无法连接成功

jxzy199306, 2016-10-21:
直接在shell运行的二进制文件，然后PC上开shadowsocks，服务器写的openwrt的IP，192.168.1.1，可以看到产生了stream，但电脑无法FQ，具体信息如下：

`root@OpenWrt:~/kcptun# /root/kcptun/client -c config.json
2016/01/31 19:03:16 main.go:250: version: SELFBUILD
2016/01/31 19:03:18 main.go:286: listening on: [::]:9000
2016/01/31 19:03:18 main.go:287: encryption: aes
2016/01/31 19:03:18 main.go:288: nodelay parameters: 1 20 2 1
2016/01/31 19:03:18 main.go:289: remote address: xxxxxxxxxx:8888
2016/01/31 19:03:18 main.go:290: sndwnd: 1024 rcvwnd: 1024
2016/01/31 19:03:18 main.go:291: compression: false
2016/01/31 19:03:18 main.go:292: mtu: 1350
2016/01/31 19:03:18 main.go:293: datashard: 10 parityshard: 3
2016/01/31 19:03:18 main.go:294: acknodelay: false
2016/01/31 19:03:18 main.go:295: dscp: 0
2016/01/31 19:03:18 main.go:296: sockbuf: 4194304
2016/01/31 19:03:18 main.go:297: keepalive: 10
2016/01/31 19:03:18 main.go:298: conn: 1
2016/01/31 19:03:18 main.go:299: autoexpire: 60
2016/01/31 19:03:53 main.go:56: stream opened
2016/01/31 19:04:03 main.go:79: stream closed
2016/01/31 19:04:04 main.go:56: stream opened
2016/01/31 19:04:13 main.go:79: stream closed
2016/01/31 19:04:18 main.go:405: session scavenged
2016/01/31 19:04:44 main.go:56: stream opened
2016/01/31 19:04:48 main.go:405: session scavenged
2016/01/31 19:04:54 main.go:79: stream closed
2016/01/31 19:05:34 main.go:56: stream opened
2016/01/31 19:05:44 main.go:79: stream closed
2016/01/31 19:05:48 main.go:405: session scavenged
2016/01/31 19:07:38 main.go:56: stream opened
2016/01/31 19:07:48 main.go:405: session scavenged
2016/01/31 19:08:08 main.go:79: stream closed
2016/01/31 19:08:08 main.go:56: stream opened
`

请问可能是什么原因呢？

kirileec, 2016-11-10:
我也一样, 也是这个路由器,同样的情况, 后面甚至换了WNDR4300, 情况依旧, 同样的参数在电脑上就可以正常使用

bettermanbao, 2016-11-11:
看log，不应该有问题。BB和CC我都用过，一点问题都没有。

kirileec, 2016-11-13:
我后面测试了下好像是ss服务端的问题, kcptun这边连接是没问题的,  我的版本是2.8.2, 如果在路由器上跑kcptun, 那么ss服务端日志里会提示加密之类的错误, 很奇怪, 是不是对服务端有一定的要求@bettermanbao

bettermanbao, 2016-11-18:
服务器端最好和客户端用相同版本

closed 2016-11-18 by bettermanbao