allenkey666/openwrt-kcptun #18: 最新的1118版本，netgear 3700v4 报错

tisswb, 2016-11-22:
通过/usr/bin/kcptun/client -c /var/etc/kcptun/client.json命令启动ar71xx这个后，只要一连接，就会报下面的错误

`
panic: runtime error: slice bounds out of range

goroutine 18 [running]:
github.com/klauspost/compress/snappy.encodeBlock(0x105d0013, 0x12ac9, 0x12ac9, 0x1059a000, 0x1a, 0x10000, 0xb900f380)
	/opt/mipsgo/src/github.com/klauspost/compress/snappy/encode_other.go:175 +0x10c4
github.com/klauspost/compress/snappy.Encode(0x105d0012, 0x12aca, 0x12aca, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/klauspost/compress/snappy/encode.go:37 +0x370
github.com/klauspost/compress/snappy.(*Writer).write(0x1053c7e0, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/klauspost/compress/snappy/encode.go:229 +0x280
github.com/klauspost/compress/snappy.(*Writer).Flush(0x1053c7e0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/klauspost/compress/snappy/encode.go:272 +0xb0
main.(*compStream).Write(0x105387d0, 0x10608000, 0x1a, 0x10008, 0x1a, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:40 +0x88
github.com/xtaci/smux.(*Session).writeFrame(0x10510190, 0x1020000, 0x3, 0x10634000, 0x12, 0x8000, 0x1050ad20, 0x0, 0x0)
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:271 +0x4b8
github.com/xtaci/smux.(*Stream).Write(0x1053a090, 0x10634000, 0x12, 0x8000, 0x12, 0x0, 0x0)
	/opt/mipsgo/src/github.com/xtaci/smux/stream.go:88 +0x190
io.Copy(0x779ea4c8, 0x1053a090, 0x779ea4e0, 0x1051c2a8, 0x0, 0x3, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:364 +0x288
main.func·002()
	/home/openwrt/kcptun/client/main.go:67 +0xd8
created by main.handleClient
	/home/openwrt/kcptun/client/main.go:67 +0x578

goroutine 1 [IO wait]:
net.(*pollDesc).Wait(0x105560b8, 0x72, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:84 +0x4c
net.(*pollDesc).WaitRead(0x105560b8, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:89 +0x48
net.(*netFD).accept(0x10556080, 0x0, 0x779e8b10, 0x105383f0)
	/opt/mipsgo/src/net/fd_unix.go:419 +0x3dc
net.(*TCPListener).AcceptTCP(0x1051c2a0, 0x1051c2b0, 0x0, 0x0)
	/opt/mipsgo/src/net/tcpsock_posix.go:234 +0x58
main.func·005(0x10500aa0, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:376 +0x2620
reflect.Value.call(0x223d60, 0x2caf14, 0x13, 0x277e68, 0x4, 0x10529ac0, 0x1, 0x1, 0x0, 0x0, ...)
	/opt/mipsgo/src/reflect/value.go:419 +0xff0
reflect.Value.Call(0x223d60, 0x2caf14, 0x13, 0x10529ac0, 0x1, 0x1, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/reflect/value.go:296 +0xb0
github.com/urfave/cli.HandleAction(0x223d60, 0x2caf14, 0x10500aa0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/urfave/cli/app.go:487 +0x27c
github.com/urfave/cli.(*App).Run(0x10566000, 0x1050a000, 0x3, 0x3, 0x0, 0x0)
	/opt/mipsgo/src/github.com/urfave/cli/app.go:245 +0x9e0
main.main()
	/home/openwrt/kcptun/client/main.go:406 +0x1418

goroutine 5 [syscall]:
os/signal.loop()
	/opt/mipsgo/src/os/signal/signal_unix.go:21 +0x30
created by os/signal.init·1
	/opt/mipsgo/src/os/signal/signal_unix.go:27 +0x58

goroutine 6 [chan receive]:
main.sigHandler()
	/home/openwrt/kcptun/client/signal.go:23 +0x150
created by main.init·1
	/home/openwrt/kcptun/client/signal.go:15 +0x48

goroutine 7 [select]:
github.com/xtaci/kcp-go.(*UDPSession).updateTask(0x1057e080)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:511 +0x33c
created by github.com/xtaci/kcp-go.newUDPSession
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:120 +0x524

goroutine 8 [select]:
github.com/xtaci/kcp-go.(*UDPSession).outputTask(0x1057e080)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:412 +0x1a60
[... 96 lines cut ...]
net.(*TCPConn).ReadFrom(0x1051c2b0, 0x779ea450, 0x1053a120, 0x779ea480, 0x1051c2b0, 0x0, 0x0)
	/opt/mipsgo/src/net/tcpsock_posix.go:72 +0x118
io.Copy(0x779ea438, 0x1051c2b0, 0x779ea450, 0x1053a120, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:358 +0x154
main.func·001()
	/home/openwrt/kcptun/client/main.go:64 +0xd8
created by main.handleClient
	/home/openwrt/kcptun/client/main.go:64 +0x448

goroutine 20 [IO wait]:
net.(*pollDesc).Wait(0x10560678, 0x72, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:84 +0x4c
net.(*pollDesc).WaitRead(0x10560678, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:89 +0x48
net.(*netFD).Read(0x10560640, 0x1064c000, 0x8000, 0x8000, 0x0, 0x779e8b10, 0x10538350)
	/opt/mipsgo/src/net/fd_unix.go:242 +0x3e0
net.(*conn).Read(0x1051c2b0, 0x1064c000, 0x8000, 0x8000, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/net/net.go:121 +0xf0
io.Copy(0x779ea4c8, 0x1053a120, 0x779ea4e0, 0x1051c2b0, 0x0, 0x3, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:362 +0x208
main.func·002()
	/home/openwrt/kcptun/client/main.go:67 +0xd8
created by main.handleClient
	/home/openwrt/kcptun/client/main.go:67 +0x578
`

bettermanbao, 2016-11-23:
use --nocomp or "nocomp": true

closed 2016-11-23 by bettermanbao

tisswb, 2016-11-23:
@bettermanbao 谢谢，我回家试试