allenkey666/openwrt-kcptun #26: client_linux_mips 启动正常，shadowsocks链接时崩溃

zixia, 2017-03-02:
版本：
* kcptun-openwrt-ar71xx-20170221.tar.gz
* D-Link DIR-505 rev. A1
* OpenWrt Chaos Calmer 15.05.1 / LuCI 15.05-149-g0d8bbd2 Release (git-15.363.78009-956be55)

启动正常，但是shadowsocks链接到kcptun端口后，出错退出。

日志如下：
```shell
2017/03/02 03:11:59 main.go:64: stream opened
panic: runtime error: slice bounds out of range

goroutine 13 [running]:
github.com/golang/snappy.encodeBlock(0x105c0014, 0x12ac8, 0x12ac8, 0x105a8000, 0x232, 0x10000, 0x7faf41a)
	/opt/mipsgo/src/github.com/golang/snappy/encode_other.go:175 +0x10c4
github.com/golang/snappy.Encode(0x105c0012, 0x12aca, 0x12aca, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:37 +0x370
github.com/golang/snappy.(*Writer).write(0x1053c810, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:229 +0x280
github.com/golang/snappy.(*Writer).Flush(0x1053c810, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/encode.go:272 +0xb0
main.(*compStream).Write(0x10538300, 0x1060a000, 0x232, 0x10008, 0x232, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:42 +0x88
github.com/xtaci/smux.(*Session).sendLoop(0x10510190)
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:302 +0x588
created by github.com/xtaci/smux.newSession
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:73 +0x394

goroutine 1 [IO wait]:
net.(*pollDesc).Wait(0x10532138, 0x72, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:84 +0x4c
net.(*pollDesc).WaitRead(0x10532138, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:89 +0x48
net.(*netFD).accept(0x10532100, 0x0, 0x77dffa68, 0x10649d70)
	/opt/mipsgo/src/net/fd_unix.go:419 +0x3dc
net.(*TCPListener).AcceptTCP(0x1051c2a8, 0x77e01350, 0x0, 0x0)
	/opt/mipsgo/src/net/tcpsock_posix.go:234 +0x58
main.func·005(0x10500a00, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:404 +0x2acc
reflect.Value.call(0x234760, 0x2e0c94, 0x13, 0x28aae0, 0x4, 0x10529a50, 0x1, 0x1, 0x0, 0x0, ...)
	/opt/mipsgo/src/reflect/value.go:419 +0xff0
reflect.Value.Call(0x234760, 0x2e0c94, 0x13, 0x10529a50, 0x1, 0x1, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/reflect/value.go:296 +0xb0
github.com/urfave/cli.HandleAction(0x234760, 0x2e0c94, 0x10500a00, 0x0, 0x0)
	/opt/mipsgo/src/github.com/urfave/cli/app.go:487 +0x27c
github.com/urfave/cli.(*App).Run(0x10568000, 0x1050a000, 0x3, 0x3, 0x0, 0x0)
	/opt/mipsgo/src/github.com/urfave/cli/app.go:245 +0x9e0
main.main()
	/home/openwrt/kcptun/client/main.go:428 +0x1660

goroutine 5 [syscall]:
os/signal.loop()
	/opt/mipsgo/src/os/signal/signal_unix.go:21 +0x30
created by os/signal.init·1
	/opt/mipsgo/src/os/signal/signal_unix.go:27 +0x58

goroutine 6 [chan receive]:
main.sigHandler()
	/home/openwrt/kcptun/client/signal.go:23 +0x150
created by main.init·1
	/home/openwrt/kcptun/client/signal.go:15 +0x48

goroutine 7 [select]:
github.com/xtaci/kcp-go.(*UDPSession).updateTask(0x10562100)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:502 +0x3c4
created by github.com/xtaci/kcp-go.newUDPSession
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:118 +0x4c0

goroutine 8 [select]:
github.com/xtaci/kcp-go.(*UDPSession).outputTask(0x10562100)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:408 +0x1a24
created by github.com/xtaci/kcp-go.newUDPSession
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:119 +0x4f0

goroutine 9 [select]:
github.com/xtaci/kcp-go.(*UDPSession).readLoop(0x10562100)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:640 +0x50c
created by github.com/xtaci/kcp-go.newUDPSession
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:121 +0x530

goroutine 11 [IO wait]:
net.(*pollDesc).Wait(0x10532cf8, 0x72, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:84 +0x4c
net.(*pollDesc).WaitRead(0x10532cf8, 0x0, 0x0)
	/opt/mipsgo/src/net/fd_poll_runtime.go:89 +0x48
net.(*netFD).readFrom(0x10532cc0, 0x1057e800, 0x800, 0x800, 0x0, 0x0, 0x0, 0x77dffa68, 0x10649678)
	/opt/mipsgo/src/net/fd_unix.go:269 +0x418
net.(*UDPConn).ReadFromUDP(0x1051c3c8, 0x1057e800, 0x800, 0x800, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/net/udpsock_posix.go:67 +0x108
net.(*UDPConn).ReadFrom(0x1051c3c8, 0x1057e800, 0x800, 0x800, 0x0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/net/udpsock_posix.go:82 +0x104
github.com/xtaci/kcp-go.(*UDPSession).receiver(0x10562100, 0x10532080)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:621 +0x108
created by github.com/xtaci/kcp-go.(*UDPSession).readLoop
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:637 +0x84

goroutine 12 [select]:
github.com/xtaci/kcp-go.(*UDPSession).Read(0x10562100, 0x105e4000, 0x4, 0x12ace, 0x8, 0x0, 0x0)
	/opt/mipsgo/src/github.com/xtaci/kcp-go/sess.go:182 +0x8d0
io.ReadAtLeast(0x77e011e0, 0x10562100, 0x105e4000, 0x4, 0x12ace, 0x4, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:298 +0x118
io.ReadFull(0x77e011e0, 0x10562100, 0x105e4000, 0x4, 0x12ace, 0x10558400, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:316 +0x6c
github.com/golang/snappy.(*Reader).readFull(0x10532140, 0x105e4000, 0x4, 0x12ace, 0x1500e60, 0x10290bc)
	/opt/mipsgo/src/github.com/golang/snappy/decode.go:108 +0x6c
github.com/golang/snappy.(*Reader).Read(0x10532140, 0x105f8000, 0x8, 0x10008, 0x77e012d8, 0x0, 0x0)
	/opt/mipsgo/src/github.com/golang/snappy/decode.go:128 +0x1bc
main.(*compStream).Read(0x10538300, 0x105f8000, 0x8, 0x10008, 0x0, 0x0, 0x0)
	/home/openwrt/kcptun/client/main.go:37 +0x5c
io.ReadAtLeast(0x77e012d8, 0x10538300, 0x105f8000, 0x8, 0x10008, 0x8, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:298 +0x118
io.ReadFull(0x77e012d8, 0x10538300, 0x105f8000, 0x8, 0x10008, 0x8, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:316 +0x6c
github.com/xtaci/smux.(*Session).readFrame(0x10510190, 0x105f8000, 0x10008, 0x10008, 0x0, 0x0, 0x0, 0x0, 0x0, 0x0, ...)
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:189 +0x104
github.com/xtaci/smux.(*Session).recvLoop(0x10510190)
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:224 +0x1d8
created by github.com/xtaci/smux.newSession
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:72 +0x368

goroutine 14 [select]:
github.com/xtaci/smux.(*Session).keepalive(0x10510190)
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:272 +0x360
created by github.com/xtaci/smux.newSession
	/opt/mipsgo/src/github.com/xtaci/smux/session.go:74 +0x3c0

goroutine 15 [select]:
main.scavenger(0x105322c0, 0x258)
	/home/openwrt/kcptun/client/main.go:441 +0x818
created by main.func·005
	/home/openwrt/kcptun/client/main.go:400 +0x2a6c

goroutine 17 [select]:
main.handleClient(0x10510190, 0x77e01350, 0x1051c2a0)
	/home/openwrt/kcptun/client/main.go:77 +0x688
created by main.func·005
	/home/openwrt/kcptun/client/main.go:424 +0x33c0

goroutine 18 [select]:
github.com/xtaci/smux.(*Stream).Read(0x1053a090, 0x10624000, 0x8000, 0x8000, 0x0, 0x77dfb9c0, 0x1051c068)
	/opt/mipsgo/src/github.com/xtaci/smux/stream.go:75 +0x6a8
io.Copy(0x77e013d0, 0x1051c2e0, 0x77e01370, 0x1053a090, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:362 +0x208
net.genericReadFrom(0x77e01388, 0x1051c2a0, 0x77e01370, 0x1053a090, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/net/net.go:404 +0xb4
net.(*TCPConn).ReadFrom(0x1051c2a0, 0x77e01370, 0x1053a090, 0x77e013b8, 0x1051c2a0, 0x0, 0x0)
	/opt/mipsgo/src/net/tcpsock_posix.go:72 +0x118
io.Copy(0x77e01388, 0x1051c2a0, 0x77e01370, 0x1053a090, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:358 +0x154
main.func·001()
	/home/openwrt/kcptun/client/main.go:71 +0xf8
created by main.handleClient
	/home/openwrt/kcptun/client/main.go:71 +0x474

goroutine 19 [select]:
github.com/xtaci/smux.(*Stream).Write(0x1053a090, 0x1062c000, 0x22a, 0x8000, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/github.com/xtaci/smux/stream.go:116 +0x6c0
io.Copy(0x77e01400, 0x1053a090, 0x77e01418, 0x1051c2a0, 0x0, 0x0, 0x0, 0x0)
	/opt/mipsgo/src/io/io.go:364 +0x288
main.func·002()
	/home/openwrt/kcptun/client/main.go:74 +0xf8
created by main.handleClient
	/home/openwrt/kcptun/client/main.go:74 +0x5a4
```

PrinnyK, 2017-03-02:
@zixia 服务器和客户端均禁用压缩试试，"nocomp":true，我通过这个办法解决了，感觉是路由器性能太弱导致的，另外我这边加速效果还不如shadowsocks裸连，vps用的是版瓦工

zixia, 2017-03-03:
@PrinnyK 非常感谢。设置 nocomp 后的确不出现数组越界错误了。但是OpenWrt内置的shadowsocks-libev链接成功后却没有调通。kcptun好像没问题了，因为我用ubuntu链接openwrt的kcptun端口已经能用了。

BTW: 后来我发现Google的TCP BBR能够有类似的效果 http://www.cnvultr.com/416.html ，目前切到BBR了，可以避免低端硬件的OpenWrt跑流量到5Mbps左右时，kcptun cpu占用100%的问题。（使用salsa20 crypt；设置为none后整体只能够再快50%左右）

PrinnyK, 2017-03-04:
@zixia 老哥稳，托你的福，版瓦工上了bbr油管速度提升至4Mbps，附上OpenVZ开bbr的方法，不过需注意的是该方法违反了版瓦工TOS[https://www.91yun.org/archives/4996](url)

zixia, 2017-03-04:
我现在直接在电脑上用 kcptun ，能到40Mbps，哈哈。不过 openwrt 还是没有搞定，你可以分享一下你的具体安装包下载地址，以及openwrt的shadowsocks及kcptun的配置详情（文件）吗？

PrinnyK, 2017-03-04:
- 服务端ssr
[https://github.com/shadowsocksr/shadowsocksr](url)

- 服务端ssr user-config.json
`{
"server_port":8388,        //端口
"password":"password",     //密码
"protocol":"origin",       //协议插件
"obfs":"plain",      //混淆插件
"method":"rc4-md5"    //加密方式
}`

- 服务端kcptun
kcptun-linux-amd64-20170221.tar.gz

[https://github.com/xtaci/kcptun/releases/download/v20170221/kcptun-linux-amd64-20170221.tar.gz](url)

- 服务端kcptun config.json
`{
 "listen": ":kcp服务端口",
 "target": "ss服务器地址:ss服务端口（不是ss本地端口）",
 "key": "密码",
 "crypt": "xor",
 "mode": "fast2",
 "sndwnd": 1024,
 "rcvwnd": 1024,
 "nocomp":true
}`

- 客户端概述
平台 小米路由器mini MT7620 ramips
系统 OpenWrt Chaos Calmer 15.05.1

- 客户端kcptun
kcptun-openwrt-ramips-20170221.tar.gz

[https://github.com/bettermanbao/openwrt-kcptun/releases/download/v20170221/kcptun-openwrt-ramips-20170221.tar.gz](url)

luci-app-kcptun_1.2.1-1_chaos-calmer_all.ipk

[https://github.com/kuoruan/luci-app-kcptun/releases/download/v1.2.1/luci-app-kcptun_1.2.1-1_chaos-calmer_all.ipk](url)

- 客户端kcptun配置文件，我在luci上填的
服务器ip及端口，
密码，
禁止压缩，
fast2，
xor加密，
sndwnd:128，
rcvwnd:128（我这能控制在90%的cpu负载，速度2Mbps上下）
其余没改，本地是默认的127.0.0.1:12948

-客户端ssr
Shadowsocks-libev for OpenWrt v3.0.3

[https://dl.bintray.com/aa65535/opkg/shadowsocks-libev/3.0.3/](url)

luci-app-shadowsocks_1.6.2-1_all.ipk

[https://github.com/shadowsocks/luci-app-shadowsocks/releases/download/v1.6.2/luci-app-shadowsocks_1.6.2-1_all.ipk](url)

- 客户端ssr配置文件，在luci上填的，too
server 127.0.0.1
port 12948
method rc4-md5
password yourpwd
插件名称 我没填
插件参数 没填X2（用的ss原版模式没上混淆）

- 客户端自动翻墙配置
自动翻墙参考资料 [https://blog.phpgao.com/xiaomi_router_shadowsocks_libev_spec.html](url)
openwrt chinadns

[https://github.com/aa65535/openwrt-chinadns](url)

luci-app-chinadns_1.6.0-1_all.ipk

[https://github.com/aa65535/openwrt-dist-luci/releases/download/v1.6.0/luci-app-chinadns_1.6.0-1_all.ipk](url)

RedSocks2 for OpenWrt [https://github.com/aa65535/openwrt-redsocks2](url)

[https://sourceforge.net/projects/openwrt-dist/files/redsocks2/0.60-04dd685/](url)

luci-app-redsocks2_1.6.0-1_all.ipk

[https://github.com/aa65535/openwrt-dist-luci/releases/download/v1.6.0/luci-app-redsocks2_1.6.0-1_all.ipk](url)

zixia, 2017-03-04:
哇，好多信息。我现在用的是一个低端的 DIR505，有点是小巧。回头我换个路由器试试。

BTW 一直有一个疑问我没弄明白，请教一下：Shadowsocks**R** 与 Shadowsocks 有什么区别？

PrinnyK, 2017-03-04:
- 摘自wiki
2015年8月22日，其作者Clowwindy在GitHub上称，警察在两日前要求他停止开发Shadowsocks并删除其所有代码。[9]之后，作者停止维护Shadowsocks，其GitHub页面已被清空。[10][11]消息传出后，许多中国和外国开发商，以及Shadowsocks用户，在GitHub中对作者表示了致谢，对项目加星标，因此在当时Shadowsocks成为了GitHub上的热门项目（Trending）。[来源请求][12]但另有消息，原作者曾作出的据称“透露中国社会现状”的发言可能遭到某些中国政府支持者的检举，从而为后来被要求撤下专案原始码的事件埋

 下伏笔，而类似的因个人网络发言不当而被检举的事件在中国大陆也“时有发生”。[13]
目前，shadowsocks并未停止维护，在Github由多位贡献者维护。
8月25日，另一个用于突破网络审查的GoAgent项目也被作者自行删除。删除后几小时之内，GitHub即遭到了来自中国的DDoS攻击。开发者[谁？]普遍认为此次攻击与中国政府有关。[14]
2015年8月28日，电子前哨基金会针对Shadowsocks和GoAgent被删除一事发表评论，对中国政府针对翻墙软件作者的打击表示了"强烈的谴责"。[5]
尽管如此，Git仓库的日志显示该项目被移除以前就有大量的复制副本，不少副本仍然有用户维护。而且各大Linux包的软件仓库（apt-get、RHEL等）均有远端代理程序和终端程序的更新以及供下载使用。Shadowsocks本身也恢复了内容，并交由多人维护。另外Shadowsocks也产生了多个分支，例如ShadowsocksR。

- ShadowsocksR主页
[https://breakwa11.github.io/](url)

zixia, 2017-03-04:
哦哦，想起来了。当时一个妹子违反了GPL协议，后来就火了。:D

看了一下功能上没啥本质区别，那我就还是ss不R了。

感谢帮忙贴的详细介绍。:)

zixia, 2017-03-21:
Update:

禁用了压缩和前向纠错功能后，终于跑起来了。可惜在DIR-505上只能支持到4Mbps左右的速度（shadowsocks也在路由器上，crypt method salsa20）

具体参数如下：

```shell
--nocomp --parityshard 0  --datashard 0
```

mengzhihoing, 2017-05-10:
是流量太大受不了了吧，我之前服务器是  -mtu 1400 -sndwnd 2048 -rcvwnd 2048 ，也是出现你这样的情况，后来改成-mtu 1350-sndwnd 256 -rcvwnd 256就正常了。需要作者@bettermanbao继续优化，感谢

bettermanbao, 2017-05-10:
极3也好，K2也好。kcptun+ss的极限速度在10Mbps左右，单上ss的极限速度在40Mbps左右。

首先，一定要开HNAT。这样才能让CPU全部用在kcptun和ss上。

ss用原版，rc4-md5加密。
kcptun，salsa20，fast mode，rcvwnd/sndwnd 256，parityshard/datashard 0。
正常优化就是这样了。

因为主要是kcptun拖后腿，所以更变态一点可以把kcptun的加密设成none，还能再快一些。反正目前没观察到有人关照kcptun协议。

shNanChen, 2017-05-10:
@bettermanbao 更极限应该用socks5代理替代ss，
但是路由器上SOCKS5转透明代理不方便。

zixia, 2017-05-10:
@bettermanbao 你测试一下 `--parityshard 10  --datashard 10`这样类似的非 0 参数，应该可以复现数组越界崩溃。

bettermanbao, 2017-05-10:
@shNanChen 
最新版的ssr支持none的加密方式，你可以试试看。

@zixia 
试过ramips没有问题，ar71xx没试过。